PaxInstruments/labwiz-board #14: Analog reference

charlespax, 2016-05-19:
What should we do with VDDA?

charlespax, 2016-05-19:
We could leave it unconnected, butI have it connected to 3.3V. It can be completely ignored. However, it someone wants to play with using an external analog reference, there is a voltage on that pin.

[image]

There is a good deal of space in the area near VDDA. We could add a footprint for a precision voltage reference and just leave it unpopulated.
[image]

electrokean, 2016-09-03:
Adding the ferrite in https://github.com/PaxInstruments/labwiz-board/issues/48 would allow someone to remove that and isolate VDDA to connect in an alternate reference.

charlespax, 2016-09-03:
I like that idea. We should probably have a ferrite bead there anyway. If there's enough room, I can also add the appropriate IC footprints on the backside of the board.

An alternative approach would be to have a voltage reference on a module board. The reference can be measured by one of the GPIO pins (they all are on an ADC) for calibration of the STM32's internal reference.

charlespax, 2016-09-09:
To keep the design simple I will not implement this. To get quality analog readings an external reference can be used to calibrate the STM32 12-bit ADC or a higher quality ADC can be connected through a digital interface. Closing issue.

closed 2016-09-09 by charlespax