PaxInstruments/labwiz-board #18: MicroUSB 4-layer error

charlespax, 2016-05-23:
The MicroUSB connector footprint is not 4-layer friendly. The milled holes for the legs are on the pads, which means they will be plated. The power layer is not pulled away from the milled holes. This, unfortunately, will result in the shorting of 3.3V and GND. When version 0.1.1 boards come in this area must be modified to correct this.
[image]

Make or find a footprint what will work with four layers.

electrokean, 2016-05-23:
Ouch!  I'm glad I've been using drills for these mounting pins, even if soldering of them isn't quite as nice.

charlespax, 2016-07-14:
I've added a feature to the PCB design to pull away the internal 3.3V layer. It would be nice to have this in the actual USB part, but Eagle seems to not have that ability.
[image]

closed 2016-07-14 by charlespax