PaxInstruments/labwiz-board #23: Cortex debug header resistors

charlespax, 2016-07-15:
The board has a Cortex debug header. There are several resistors connected to this circuit. Do we need these? Are they the right values?

Also, I've placed USART1 RX and TX on two of the unused pins.

[image]

The Olimecino board does not have these resistors.
[image]

electrokean, 2016-07-15:
SWDIO actually has an internal pull-up, and SWDIO a pull-down per the STM32F103 reference manual, so I don't think these external bias resistors are mandatory.  It also states that the JTAG spec recommends the appropriate pull-up/downs, but the embedded ones remove that requirement (see 31.4.3).
Interestingly, 31.8.1 says a 100k pull-up  on SWDIO is recommended by ARM, but again probably non necessary as the STM32F103 includes these.
The series resistors are probably included on some designs in case the SWJ pins are reconfigured (via SWJ_CFG bits) to GPIOs by the application code while still connected to a programmer.

charlespax, 2016-08-08:
I'll leave the resistors in for now and ice box this issue. I'll look into it more to confirm we can remove them.