PaxInstruments/labwiz-board #24: SD card resistors and inductor

charlespax, 2016-07-15:
Are the resistors on the SD card the correct values? We use 10k resistors.
[image]

The Olimexino board uses 100k resistors. They also use an inductor. Should we have an inductor?
[image]

electrokean, 2016-07-15:
The SD card physical layer specification doesn't seem to require any pull-ups, but they're clearly a good idea to avoid floating inputs on the STM32.  I've used the Olimexino resistor configuration on my some of own designs without issue.  Interestingly I just realised I think I left them off on a recent prototype based on the STM32F103, and didn't notice any issue.
The inductor on the SD card power supply and bulk capacitance is a good idea to reduce fluctuations on the 3.3V rail during SD card operations, which can draw surges of current for short periods (write or erase).  This provides filtering for the upstream 3.3V (e.g. ADC reference) as much as for the SD card itself.
An 0805 ferrite and lower value capacitor may be OK when there is a sufficiently low impedance path from the power source, but could introduce problems when the battery is getting low, or device is powered off USB without a battery inserted, leading to data corruption or even card failure.

charlespax, 2016-08-23:
From http://www.mikrocontroller.net/attachment/21920/SDHC_SDM04G7B7_08G7B7.pdf:
[image]

From https://www.kingston.com/datasheets/SDCIT-specsheet-64gb_us.pdf:
[image]

From http://www.nxp.com/documents/application_note/AN10911.pdf:
[image]

From http://downloads.amilda.org/MODs/SDCard/SD.pdf:
[image]

From http://www.digitalspirit.org/file/index.php/obj-download/docs/sd/SdCard-MMC-Controller_From_Intel.pdf:
[image]

From https://www.renesas.com/en-in/products/software-tools/software-os-middleware-driver/dsp-numerical-software/wireless-lan--rtk5563000f00000sr.html:
[image]

From http://www.analog.com/media/en/technical-documentation/application-notes/EE335_rev1.pdf:
[image]

From http://learn.parallax.com/tutorials/language/spin/propeller-board-education-project-builders/sd-card-data-storage:
[image]

From http://www.lydiard.plus.com/hardware_pages/sd_card.htm:
[image]

From http://justanotherlanguage.org/content/tutorial_sd_card:
[image]

See also

- http://electronics.stackexchange.com/questions/83263/micro-sd-card-power-circuit-failure

charlespax, 2016-09-09:
Here is another reference design (http://electronics.stackexchange.com/questions/83263/micro-sd-card-power-circuit-failure).
[image]

Since things are working now I'm not going to implement this in revision 0.3, but will in a future rev.

charlespax, 2017-01-28:
I've added 10k pullups to the CMD and data lines. Caps don't seem to be very popular, so I'll just skip those.
[image]
[image]

commit baf4f28f (2017-01-28): Added pullups to the SD interface. Closes #24.
closed 2017-01-28 by charlespax