PaxInstruments/labwiz-board #26: Serial port connections

charlespax, 2016-07-16:
Do we need pullup resistors on the serial lines? Possibly so for the bootloader. See [AN2606 p. 26](http://www.st.com/content/ccc/resource/technical/document/application_note/b9/9b/16/3a/12/1e/40/0c/CD00167594.pdf/files/CD00167594.pdf/jcr:content/translations/en.CD00167594.pdf).
[image]

electrokean, 2016-07-16:
Always want one on the Rx pin at each device that may have its input disconnected from the transmitter

charlespax, 2016-08-08:
Done

charlespax, 2017-01-23:
The resistors were removed at some point. I'm adding them again.
[image]

charlespax, 2017-01-23:
Added back on the board.

[image]
[image]

commit 80c86e76 (2017-01-23): Added 100k pullup resistors to the USART1 lines. Closes #26.
closed 2017-01-23 by charlespax