PaxInstruments/labwiz-board #27: USB connection for bootloader

charlespax, 2016-07-16:
What do we need for enabling the bootloader to flash over USB? See [AN2606 p. 26](http://www.st.com/content/ccc/resource/technical/document/application_note/b9/9b/16/3a/12/1e/40/0c/CD00167594.pdf/files/CD00167594.pdf/jcr:content/translations/en.CD00167594.pdf).
[image]

electrokean, 2016-07-16:
STM32F103 doesn't have built in USB DFU bootloader, so it would be dependant on a separate bootloader like https://github.com/rogerclarkmelbourne/STM32duino-bootloader
See the notes there in README.md as there are a few implementation options for the disconnect feature

charlespax, 2016-08-08:
Moved to firmware repo.

electrokean, 2017-01-23:
For the more modern chips with USB DFU, you should just need a way to control BOOT0 (and possibly RESET) to start the bootloader.

If you have a spare I/O pin, you can do a nifty firmware trick by connecting that to BOOT0 along with a small capacitor.  To enter the bootloader, the firmware pulls the I/O pin high to charge the cap, and then does a reset while the cap holds BOOT0 high (slowly discharging through a pulldown).

Ref: http://stackoverflow.com/questions/26891432/jump-to-bootloader-in-stm32-through-appliction-i-e-using-boot-0-and-boot-1-pins

charlespax, 2017-01-23:
PC1 is currently not used and is a good candidate. It could easily be routed over.
[image]

I think this circuit would work. I would just connect PC1 to BOOT0 and add a cap across the 10k pulldown resistor. Not sure what value.
[image]

I would have to change the BOOT1 circuit. Right now it's normally high, which would make the device boot from SRAM. Making BOOT1 normally low would have the system default to system memory (hardware bootloader) if BOOT0 is high.
[image]

charlespax, 2017-01-23:
Since at lease some chips require a 1.5k pullup on USB_DP, we should implement the circuit. Here is an except from [AN2606](http://www.st.com/content/ccc/resource/technical/document/application_note/b9/9b/16/3a/12/1e/40/0c/CD00167594.pdf/files/CD00167594.pdf/jcr:content/translations/en.CD00167594.pdf)
[image]

charlespax, 2017-01-23:
Actually, the board already has the 1.5K pullup. I think the pullup is to indicate to the host if the device is full-speed or high-speed (see also #55 ).
[image]

charlespax, 2017-01-23:
Yes, the 1.5k pullup on DP is to indicate full-speed. It is also the same circuit to indicate high-speed.

Change made to the BOOT1 switch.
[image]

PC1 connected to BOOT0.
[image]

Capacitor added in parallel with the BOOT0 pulldown.
[image]

charlespax, 2017-01-23:
Done. Closing issue.

closed 2017-01-23 by charlespax