PaxInstruments/labwiz-board #31: Oscillator component choice

charlespax, 2016-07-22:
In the design I've made the assumption that the main oscillator and the RTC oscillator are available in the same footprint. I have to verify this or possibly change to parts that actually exist and are available.

[image]

electrokean, 2016-07-22:
Yeah, this is probably not the case.
The RTC crystal specs can be quite particular around capacitance and ESR.
Check out AN2867 - it has lots of info plus lists of some suggested parts, and it will also point out some potential issues with your layout around that area.
Finding parts that meets your price/footprint/availability requirements can be a bit of a challenge.

charlespax, 2016-08-08:
It looks like small 8 MHz crystals are pretty easy and cheap to get. A small 16 MHz device would have to be a more expensive and rare MEMS device. We'll go with 8 MHz.

Here is a commonly available component from the HQB market. When buying twenty pieces I paid 2 RMB each.
[image]

charlespax, 2016-08-08:
Here is the new crystal on the PCB. [image]

closed 2016-08-08 by charlespax