PaxInstruments/labwiz-board #39: Purpose of VBUS input

charlespax, 2016-08-08:
We have VBUS input on PA9, but I'm not sure why. We're pretty strapped for pins, so it would be nice to know if we can repurpose this pin. What is the use case here?
[image]

electrokean, 2016-08-11:
I'm guessing this is useful for knowing when a USB cable is plugged in - either for mcu based battery charger control or to begin USB enumeration.  Most pins are 5V tolerant, but I still tend to include a resistive divider myself...

charlespax, 2016-08-15:
The USB data lines handle enumeration. I've removed and repurposed the VBUS line as the VBAT_DIV2 battery voltage input.

closed 2016-08-15 by charlespax