PaxInstruments/labwiz-board #41: ESP8266 connections

charlespax, 2016-08-15:
A header is on the board for an ESP8266 module, which runs on USART1. I need to review and implement any necessary pullups/pulldowns and any other components necessary.
[image]

[image]

charlespax, 2016-08-15:
I dropped a 10 uF cap on the power lines. Everything else seems to be in order.
[image]

charlespax, 2016-08-15:
Done. Looks good.

charlespax, 2016-09-11:
WIFI now has a dedicated USART. Review connections again.

samchoy88, 2016-12-10:
Using USART3 right now
[image]

closed 2016-12-11 by charlespax