PaxInstruments/labwiz-board #42: Power output

charlespax, 2016-08-16:
I want the board to be able to output as much power as possible. I may need to select another 3.3V regulator.

- What can we draw from the battery?
- What can the 3.3V regulator output?

electrokean, 2016-09-03:
If you gave up one of the ground pins on the modules, you could route VBAT or VBUS to them.

Or just allow for a pad on one of these for people wanting to hack to easily wire a jumper across to a module.

charlespax, 2016-09-03:
It turns out the CAN and USB use the same memory and cannot be used together. Thus, I'm dropping CAN from the interface. Not a big deal. This frees up two pins. The fourteen pin connector will have the following pins.

- 3V3
- 3V3
- VBUS
- GND
- GND
- SPI_MISO
- SPI_MOSI
- SPI_SCK
- GPIO1
- GPIO2
- USART_RX
- USART_TX
- I2C_SD
- I2C_SCL

With the addition of VBUS to the module a design could draw directly from USB power or from the battery. Anything beyond that and a design could have an independent power input.

charlespax, 2016-09-09:
Each connector has a pin for VBUS. Closing issue.

closed 2016-09-09 by charlespax