PaxInstruments/labwiz-board #45: 4.7 uF cap

charlespax, 2016-08-17:
The stm32f103 datasheet indicates Vdd3 needs a 4.7 uF cap.
[image]

charlespax, 2016-08-20:
I've added the 4.7 uF cap.
[image]

It also looks like I can remove all the 10 uF caps and change the VDDA cap from on 100 nF to one 10 nF cap and one 1 uF cap.
[image]

charlespax, 2016-08-22:
Removed 10uF caps on VDD pins. Changed VDDA caps.
[image]

closed 2016-08-22 by charlespax

electrokean, 2016-08-22:
Maybe add a small ferrite bead or low value resistor between VDD and VDDA to filter noise from CPU core?

charlespax, 2016-08-23:
Good call. I've opened an issue. https://github.com/PaxInstruments/labwiz-board/issues/48