PaxInstruments/labwiz-board #50: STM32 LQFP64 compatability

charlespax, 2016-09-03:
One of the nicest aspects of the STM32 family is that the chips are very compatible. If we carefully select the peripheral pin mapping, switching between chips will be relatively trivial.

Here is how pins are mapped in LabWiz v0.2. 
[image]

If we change to this pin map, we can swap out the stm32f103rgt (1024k flash, 96k RAM) with the stm32f412rgt (1024k flash,256k RAM).
[image]

Here is the stm32f412.
[image]

charlespax, 2016-09-03:
Here's what I have right now. The unassigned pins would be for the six GPIO pins for the modules. The two SPI_NSS pins plus the six GPIO are the eight GPIO pins required by the interface.

[image]

PA10 is unused in this design and is reserved and would be used for USB_OTG if the stm32f412 were used. PB11 is also not used. The stm32f412 uses this pin for a power capacitor. That is also what compelled the decision or use only a single I2C interface.

electrokean, 2016-09-03:
Out of interest, what did you use to generate those pinout images?

charlespax, 2016-09-03:
I used [STM32CubeMX](http://www.st.com/content/st_com/en/products/development-tools/software-development-tools/stm32-software-development-tools/stm32-configurators-and-code-generators/stm32cubemx.html) from ST. It's a super flipping awesome GUI for defining all the pins and peripherals.

[image]

It will help define all the timing.
[image]

Then it will generate all the startup code in a project. It supports several IDEs. We've been using a secondary Python script to translate from SW4STM32 to and Eclipse project.
[image]

electrokean, 2016-09-03:
Thanks @charlespax. I suspected it was from STMCube.  I played with an older version, but only briefly.  I have been writing my own startup code so far.  Will have to try this latest version out, especially with PCBs for a new design about to land.

charlespax, 2016-09-05:
Given the following criteria,

- STM32 family
- LQFP64 package
- Have USB capability
- Have SDIO interface

The following twenty-seven parts are potentially compatible.

- STM32F103RC
- STM32F103RD
- STM32F103RE
- STM32F103RF
- STM32F103RG
- STM32F205RB
- STM32F205RC
- STM32F205RE
- STM32F205RF
- STM32F205RG
- STM32F215RE
- STM32F215RG
- STM32F401RB
- STM32F401RC
- STM32F401RD
- STM32F401RE
- STM32F405RG
- STM32F411RC
- STM32F411RE
- STM32F412RE
- STM32F412RG
- STM32F415RG
- STM32F446RC
- STM32F446RE
- STM32L151RD
- STM32L152RD
- STM32L162RD

There are a few down sides with taking SDIO as a criteria. The chip with the lowest amount of flash has 128k flash, 64k RAM. That's fine for development, but limits the options for reducing cost in a final product. However, if we were to use a SPI interface instead, we will be compatible with sixty chips. We'd would then only have one SPI interface for all four modules. Not a good trade off.

charlespax, 2016-09-05:
Here's where I am.

[image]

All module GPIO pins are capable of GPIO, external interrupt, and 12-bit ADC measurements. Additional features are carried on the pins as listed below.

- MOD1_GPIO1 SPI1_NSS, DAC_OUT1
- MOD1_GPIO2 PWM
- MOD2_GPIO1 GPIO PWM
- MOD3_GPIO1 SPI2_NSS

charlespax, 2016-09-09:
I've routed a new board revision. Closing this issue.

[image]

closed 2016-09-09 by charlespax