PaxInstruments/labwiz-board #57: MCP73831 PROG resistor

charlespax, 2016-09-11:
Check the PROG resistor on the battery charger. Make sure the value defines the correct charge rate.

samchoy88, 2016-12-10:
Since we face the same issue in T400 circuit design.
https://github.com/PaxInstruments/t400-electronics/issues/163

We should follow the T400 electronic design and use 2.32K (1%) in here
With a 2.32K resistor the typical charge current would be 1000 / 2.32 = 431mA
The highest and lowest possible charge currents, using 10% for the chip and 1% for the resistor would be:
Maximum 479mA
Minimum 384mA

[image]

charlespax, 2016-12-11:
I've added a 2.32K resistor to the RLC library and added it to the schematic.

[image]

closed 2016-12-11 by charlespax