PaxInstruments/labwiz-board #59: USB power protection

charlespax, 2016-09-11:
What protection is needed on the USB power? Right now it's connected directly to VBUS.

robojay, 2016-10-30:
Add a PTC resettable fuse between the USB connector VCC pin and VBUS.

I've used Bel Fuse 0ZCJ0050FF2G (Digi-Key 507-1802-1-ND) in similar applications.

robojay, 2016-12-11:
Add a transient voltage suppressor between VUSB and ground.  Something like SMBJ5.0A will work.

charlespax, 2017-01-23:
I've added a PTC resettable fuse. I'm going for the 0805 package because that fits best on the board.  This part seems good: http://www.digikey.com/product-detail/en/bel-fuse-inc/0ZCK0050FF2E/507-1813-2-ND/4156141
[image]
[image]

charlespax, 2017-01-23:
I've added a transient voltage suppressor to USB_VBUS. It's the same as what I have on the data lines.

[image]
[image]

commit 2660a11f (2017-01-23): Added transient voltage suppressor to USB_VCC. Closes #59.
closed 2017-01-23 by charlespax