PaxInstruments/labwiz-board #103: RTC battery size

charlespax, 2016-09-26:
Maybe use a smaller RTC battery. First, let's find out how long we want it to last. Then let's find out what battery meets that requirement. A super cap would be great.

electrokean, 2017-01-25:
I've used a small rechargeble lithium batteries like the one on the Seeed modules mentioned in #16 on a Rasp Pi hat with RTC, but haven't done any testing on how long it lasts.
They are Seiko MS621FE 3V 5.5mAh
http://www.sii.co.jp/en/me/datasheets/ms-rechargeable/ms621fe/
I originally bought these to repair some portable barcode scanners for a client.
Being Lithium, even though ridicuously tiny, they still have shipping restrictions.

Elna make a similar size (actually a bit smaller) Supercap in the DSK series, 3.3V 0.22F.
I got some surplus stock which appear to be obsolete now, but there is a replacement worth a look.
http://mouser.com/ProductDetail/Elna/DSK-3R3H224U-HL/

With the main device battery providing power normally and charging these as secondary sources, I don't think they really need a long run time as these are basically there just to keep the RTC running to avoid having to reset the RTC.  With the potential for wireless, the device could even be made to sync time over NTP.

charlespax, 2017-01-25:
It would be nice to use a technology with no shipping restrictions. This way I can sell the PCBA as cheaply as possible. That's basically a super cap or a LiPo battery that is not soldered to the board.

electrokean, 2017-01-25:
Agreed.  The DSK supercap I mentioned above seems like a good option, is readily availble, and takes up little room.  The alternative is probably a CR1220 holder, but not sure you have space for that.

I did discover this thread about similar supercaps leaking/faling in Fluke meters on EEVblog Forums
http://www.eevblog.com/forum/testgear/fluke-189-with-leaking-surface-mount-coin-cell/
Interesting to note that the final post on that thread indicates Fluke apparently switched from these supercaps to the tiny Seiko lithium batteries I mentioned above.

There are a few mentions there of different style (larger) of supercap which might be a safer bet.  The Panasonic EEC-S0HD334H is another one I have, is cheaper, but is also through hole and rated for 5.5V (probably a smaller 3.3V version also).

I just noticed on DigiKey that there is an Elna DCK series supercap, which appears to have a compaible footprint to the Seiko MS621 rechargeable lithium batteries.  Maybe good if you have limited space, and not sure which is best.  You could even have more than one footprint (SMT+THT) to allow end user to replace.

Like all component selection, too many choices all with various tradeoffs.

charlespax, 2017-01-25:
The CR1220 is what's on the T400 and what has been on the LabWiz board. It doesn't quite fit on this board now that I've put the ESP-09 module in its former location.

A CR1025 battery will fit on the board. Here's the datasheet for the holder http://www.memoryprotectiondevices.com/datasheets/BK-870-datasheet.pdf.
[image]

charlespax, 2017-01-26:
I've put the RTC super cap idea (#16) on ice. We can explore that again in the future.

charlespax, 2017-01-28:
Th3e 1025 battery appears to work out nicely for layout. Closing issue.

closed 2017-01-28 by charlespax