PaxInstruments/labwiz-board #111: USB connector interference

charlespax, 2016-09-26:
The USB connector interferes with the enclosure. Find a more suitable connector. Maybe move the connector.

charlespax, 2017-01-22:
There does exist a better connector. In the HQB markets I've seen a flangeless version of the connector in the T400. It looks more or less like this...
[image]

I have to find the part and implement the footprint. I believe one of the Blinkin Labs products uses it. https://github.com/Blinkinlabs

electrokean, 2017-01-22:
Looks like the same footprint at the Molex 105017-0001 or FCI 10118194-0001LF
I've found them much less likely to be pulled off the PCB than the flanged versions

electrokean, 2017-01-22:
Hang on - isn't this the footprint you referenced in https://github.com/PaxInstruments/labwiz-board/issues/18

charlespax, 2017-01-23:
Nope. #18 is about the footprint and the internal layers. This issue (#111) is about the flange of the MicroUSB connector interfering with the enclosure.

This part is similar to what is on the T400. Check out the flange.
[image]

electrokean, 2017-01-23:
Aha, I see.  Not the mounting flange, but the "entry" flange.
Fot the Molex part, that would be 105017-1001
http://www.digikey.com/product-detail/en/molex-llc/1050171001/WM11262CT-ND/3313406