PaxInstruments/labwiz-board #76: USB protection diodes

charlespax, 2016-09-26:
We are using some diodes on the USB lines. Make we are using the right diodes for this. Make sure we are using the right footprint.

robojay, 2016-10-30:
Take a look at this app note from ST.  It covers ESD protection and not needing the series resistors you have in the current design.

[ST AN4879.pdf](https://github.com/PaxInstruments/labwiz-board/files/560836/ST.AN4879.pdf)

charlespax, 2016-11-10:
Nice. We should ditch the resistors then. I also want to look at high-speed devices because the PCB will be compatible with several STM32Fx families.

[image]

charlespax, 2017-01-27:
Series resistors removed.
[image]
[image]

charlespax, 2017-01-27:
- [ ] "The full speed USB driver impedance on STM32 MCUs is guaranteed by design to be between 28 ohms and 44 ohms. The result is that these products do not need an external resistor on the board for the output driver impedance."
- [ ] "The track routing should respect 90 ohm +/-15% differential from the STM32 MCU’s package to the receptacle. It is important to route the two data lines in parallel to ensure the best impedance matching. A 45 ohm +/-15% single ended trace is recommended."
- [ ] "Routing USB signals using a minimum of vias and corners reduces the signal reflections and the impedance changes. When it becomes necessary to turn by 90°, it is preferred to use two 45° turns or an arc instead of making a single 90° turn. This action reduces the reflections on the signal by minimizing the impedance discontinuities."
- [ ] "A protection against ESD and EMI is needed... The HBM requires that the USB pins of the component device are tolerant to up to 2 kV discharge, this is the case for the STM32 MCUs." Done. The ESD diodes can handle the 2 kv.
[image]
- [ ] "The component needs to be placed as close as possible to the receptacle to add ESD protection against high ESD surge. The USBLC6 component is recommended for such protection on VBUS and USB data lines."
- [ ] "It is recommended to implement a resistor bridge to lower the voltage used for VBUS detection. This recommendation is even more important when VDD on the STM32 MCU is not present while VBUS is supplied from the connector." Done.
[image]

commit c754c443 (2017-01-27): PC4 is now VBUS_DETECT. Removed unnecessary USB data series resistors. Closes #76.'
closed 2017-01-27 by charlespax