PaxInstruments/labwiz-board #118: Power down completely

charlespax, 2016-11-10:
Does LabWiz ever power down completely or just go into a deep sleep mode? There are no power switching components.

Here is a power switching example.
[image]

charlespax, 2016-11-10:
It has been suggested that we disconnect the battery with a MOSFET.  By completely disconnecting the battery via MOSFET you don't have any leakage current (e.g. the battery sense divider, quiescent current of regulator, etc.). The tradeoff is complexity though. There may be a part that integrates all this functionality

robojay, 2016-12-11:
Related items... The 3.3V regulator dropout voltage (MIC5219-3.3, max of 400mV at 150mA, or max of 600mV at a 500mA burst) should be taken into account with battery voltage state.  How low do you want to run the battery? 3.3V + 0.4V = battery needs to be >= 3.7V, etc.

If you don't have a way to disconnect the battery, you will continue to run it down, potentially inflicting permanent damage to the battery.

robojay, 2016-12-11:
Depending on your overall current requirements, there are parts around that integrate charging, battery disconnect, something like this: http://www.microchip.com/wwwproducts/en/MCP73871

charlespax, 2017-02-02:
Li-Po discharge curve:
[image]

The Qduino (https://www.sparkfun.com/products/13614) uses the AP2112K-3.3TRG1 (http://www.diodes.com/_files/datasheets/AP2112.pdf).

AP2112K output voltage vs input voltage:
[image]

I think this part will take care of us pretty well. We can get our 3.3V all the way down to a battery voltage of 3.3V.

charlespax, 2017-02-02:
The Qduino also has autoswitching between battery and USB:
[image]

I've incorporated that design:
[image]