PaxInstruments/labwiz-board #125: SPI pullup/pulldown resistors

charlespax, 2017-01-22:
Do we need any pullup or pulldown resistors on the SPI lines?

electrokean, 2017-01-22:
Generally I'd put a pullup on the SPI !CS signal at the slave.  I may also put a pullup on the MISO at the master, but this can usually be done via the internal I/O pin configuration.  An external (lower value) may be needed if there are long connecting cables.
A reference to Arduino SPI best practices: https://www.dorkbotpdx.org/blog/paul/better_spi_bus_design_in_3_steps

charlespax, 2017-01-27:
In the STM32 datasheet and reference manual I cannot find anything indicating that SPI has any pullup resistors. However, GPIO does. I'm not sure if that also translates to that pin in SPI mode. I'll just keep them for now.

I'm putting this in the ice box for future investigation.

charlespax, 2017-01-27:
I'm putting pullups on the board. We can try them.
[image]

charlespax, 2017-01-28:
Moved to Milestone 4.1 for review after we have the 0.4 boards.