PaxInstruments/labwiz-board #127: Hardware for USB bootloader?

charlespax, 2017-01-22:
See https://github.com/PaxInstruments/PaxInstruments-LabWiz-firmware/issues/3

electrokean, 2017-01-22:
Maybe time to re-evaluate use of STM32F103 and use something newer like the STM32L476
Lower power, faster, better features, proper USB DFU support - but I haven't compared pinout...
Maybe check it out in STM32Cube?
There's active development for Arduino support https://github.com/GrumpyOldPizza/arduino-STM32L4

charlespax, 2017-01-23:
I went through a bunch of chips to make sure the layout is compatible with as many chips as possible. See https://github.com/PaxInstruments/labwiz-board/issues/50. STM32L476 is not on the list, but I'm not sure why. Was that chip released after September 5th?

I just went through CubeMX and all the pin mapping appears to work out.

STM32L476:
[image]

Current LabWiz STM32F103:
[image]

electrokean, 2017-01-23:
Great!  The L4 series look really nice.
I believe it was released about 12 months ago, but it may not have been added to STMCube32 till later.
I only started looking at it recently, although I'm more interested in its little brother the L432

charlespax, 2017-01-24:
I think we're good on the bootloader hardware. Closing...

closed 2017-01-24 by charlespax