PaxInstruments/labwiz-board #133: Put USART3 on modules

charlespax, 2017-01-24:
Right now all four modules share USART2 while USART3 is dedicated to the wifi modules. Maybe we should have modules 1 and 2 share USART while modules 3 and 4 share USART3.
[image]

electrokean, 2017-01-24:
If you're going to share the USART between module slots, can I suggest using AND gates on the receive pins to avoid electrical contention.  Something like the 74LVC1G08 or 74LVC2G08.  Else, I'd suggest only connecting two module slots.

charlespax, 2017-01-25:
Good call. Do you mean AND gate or OR gate?

I think we would want a four input OR gate (one for each module TX) with the output going into STM32 RX.

The SPI MISO lines should also be on OR gates unless SPI already handles this. I2C should be fine since it's made to handle the electrical contention issue.

charlespax, 2017-01-25:
Maybe just kill the idea of serial on all modules. We can take the STM32 RX and TX pins and use them as GPIO. If we route them both to either modules 2 or 4, then that module can have an extra capability

electrokean, 2017-01-25:
I did mean AND gate.  The UART signals idle in a high state, so are active low.  An AND gate can be used to combine the negative logic.  I was thinking you'd use two of the tiny 2 input AND gates, to combine the RX signals coming from each pair of modules into USART2 and USART3 repectively.

The idea would be that the modules would act in a kind of multi-drop bus architechture.  You could use a GPIO or some address jumpers on each module to make them unique.

If you're going to keep the ESP8266, then I guess you only have USART2 available?  In that case you'd need a 4-input AND gate.  No such thing in the TinyLogic families, but you could use two 3-input ANDs e.g. 74LVC1G11 or NC7SZ11.  The SOT363/SC-70-6/SC-88 packages don't take up much space.

Regarding SPI, the DO lines on well behaved SPI slaves should tri-state.  Check the link I posted in #125 re SPI best practices.  If an SPI slave device on a module is not well behaved, the module should implement a separate tri-state buffer.